mainsail-crew/mainsail #1257: Pin files to top of gcode files list.

PhilBaz, 2023-02-07:
### Requested feature:

Would be nice to have the ability to pin frequently used files to the top of the gcode files list. Then maybe an option to have this be applied on the the list page, the preview on the main page, or both.

### Solves the following problem:

Hunting...

### Additional information:

Sorry for all the requests! Looking forward to the next update :)

GAZ082, 2023-05-29:
As discussed in another ticket, if the width real state is a concern, this can be added to the right click menu and color code the file name or make a pin visible.

meteyou, 2023-05-29:
i think this is a good option, instead to have a own column for the pins, to safe space on the screen.

4rnoP, 2023-12-11:
Hi,
I'm new to the mainsail project and would like to implement this feature.
However, I'm not sure what would be the optimal way of saving the pinned files. To my understanding, the only way to save persistent information is to save it with the database API of moonraker, maybe under something like `gcodefiles.pinned` under the mainsail namespace, and it would contain the list of pinned items ?

Any guidance would be appreciated :)

meteyou, 2023-12-11:
@4rnoP Thank you for your interest! Yes, you are right. You have to write the files (at least the names or IDs) to the Moonraker DB. I would name the namespace `gcodefiles.pinnedFiles` to be more precise.

Just as a further idea:
I suggest adding a 4th tab to the status panel. But since it is already tight in some languages, you should replace the words with icons at the same time. You should also pay attention to this PR (https://github.com/mainsail-crew/mainsail/pull/1678). I will merge it in the next few days. Currently, I still have to finalize the light-theme PR.

We only have these current docs for contributors: https://docs.mainsail.xyz/overview/development

PhilBaz, 2025-02-20:
Just giving this a bump...would be great to see implemented. @4rnoP Keep going! :)

PhilBaz, 2026-01-15:
Just want to say that  @4rnoP  came close to finishing this feature. https://github.com/mainsail-crew/mainsail/pull/1684 maybe someone can continue on from the work they did? seemed pretty close to done. 

Hope to see this one day as I agree it would be a nice feature.