mainsail-crew/mainsail #1262: Update OverlayScrollbars to v2

KingSora, 2023-02-10:
### Requested feature:

Currently [mainsail](https://github.com/mainsail-crew/mainsail) uses `v1` of [OverlayScrollbars](https://github.com/KingSora/OverlayScrollbars). `v2` was released recently and has a lot of improvements compared to `v2`. (Focus on modern browsers, 50% smaller size, treeshaking, deferred initialization and many more).
A full list of features and changes can be found [here](https://github.com/KingSora/OverlayScrollbars/blob/master/packages/overlayscrollbars/CHANGELOG.md).

meteyou, 2023-03-26:
@KingSora thx for your FR. I have now checked overlayscrollbars and wanted to update it. But overlayscrollbars-vue is only compatible with vue3 since v0.3.0. Currently, Mainsail still uses vue2. Am I missing something?

KingSora, 2023-03-26:
@meteyou the newest overlayscrollbars-vue package is indeed only compatible with vue3. But since all the framework wrappers (vue, react etc.) are just tiny wrappers around the original library there can also be a vue2 version for it. I could write something like that if youre interested

meteyou, 2023-03-26:
@KingSora a small example would be nice! thx!