mainsail-crew/mainsail #1265: after loading history - there are already 2 selected jobs

fused3dprints, 2023-02-13:
### Mainsail Version:

v2.4.1

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I just noticed that i automatically have 2 selected jobs whenever i load history tab but the details are now shown.
I can export the jobs (via CSV) and can confirm these are old jobs i printed but have deleted them long time ago.

[image]
[print_history.csv](https://github.com/mainsail-crew/mainsail/files/10722385/print_history.csv)

### What did you expect to happen instead?

my other printer does not have any "pre-selected" jobs when selecting the history tab

### How to reproduce this bug?

i just need to open the history tab

### Additional information:

moonraker.log shows 2 Invalid job uids:

utils.ServerError: Invalid job uid: 0000E5
utils.ServerError: Invalid job uid: 000103

[moonraker.log](https://github.com/mainsail-crew/mainsail/files/10722371/moonraker.log)

meteyou, 2023-02-14:
pls post your output of http://<ip or hostname>/?namespace=mainsail

closed 2023-03-25 by github-actions[bot]

Luoti, 2024-01-23:
I had this problem too. Solved it by deleting the `selectedJobs` from the database with curl:

```bash
curl -X DELETE "http://[IP_OF_THE_PRINTER]/server/database/item?namespace=mainsail&key=view.history.selectedJobs"
```