mainsail-crew/mainsail #1279: 3D Gcode viewer not opening on Desktop

Edigorin, 2023-02-19:
### Mainsail Version:

2.4.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

3D Gcode Viewer not opening when clicked

### What did you expect to happen instead?

3D Gcode Viewer opens

### How to reproduce this bug?

Click on the 3D Gcode tab on the left

### Additional information:

After clicking "3D Gcode Viewer" tab on the left, the browser stays on the current page (Dashboard for example), and becomes totally unresponsive. The page has to be closed and opened again to become usable.
Issue happens on Chrome and Edge on Windows 11 Desktop Computer, but not on Android Chrome. Happens with 2 different 3D Printers.

meteyou, 2023-02-19:
you have to wait some seconds, because the gcode viewer will be loaded from the host. this part is not in the init package.

closed 2023-02-19 by meteyou

Edigorin, 2023-02-20:
I've waited more than 30min once, still unresponsive. It works fine on mobile, loading after a few seconds. But on Desktop, the page doesn't load at all. It's not only that I don't see the Gcode, I don't see any page elements at all (3D bed, options,...)

meteyou, 2023-02-20:
then check your browser log for more details. maybe you have a adblocker or something similar which blocks the gcode viewer