mainsail-crew/mainsail #1281: Turnable Thumbnail

Ringo80X, 2023-02-20:
### Requested feature:

I think it would be helpful if we could rotate the thumbnail of the currently printed part.

meteyou, 2023-02-20:
I can't think of a use case for it... What would be the advantage of having the image upside down?

DVSVIDEO, 2023-02-21:
I think what the user meant here is that the small stamp image is a 3D image and he wants to rotate it in space.
He doesn't know that the slicer only "takes" a tiny picture of the model and puts this tiny picture into the .stl file.
I suggest you switch to the 3D g-code viewer tab and press the local file button there and then you can rotate the model for printing.

meteyou, 2023-02-22:
Ahh ok! Now I understand this FR.

No this is not possible, because it's just an image, which will be created from the Slicer. (as @DVSVIDEO already described)

Matszwe02, 2023-02-26:
In the other hand, there could be something like automatic thumbnail generation, like gcode viewer, but only for making that 3d thumbnail. It will be still a nice idea, visible cross section, etc.

meteyou, 2023-02-26:
the gcode viewer needs the most performance of the complete mainsail interface. so this is really overkill for this function.

closed 2023-02-26 by meteyou