mainsail-crew/mainsail #1283: I think new pause @ layer isn't working properly?

PhilBaz, 2023-02-22:
### Mainsail Version:

2.5 beta

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I started with a comment here. Sorry if that's the wrong place to comment.
https://github.com/mainsail-crew/mainsail/pull/1230

When I click the top input field (that's where I expect to enter the layer?) I get a box that says no data. I cant enter anything into it.

first
[image]
then
[image]

### What did you expect to happen instead?

Have the ability to set a layer number there.

### How to reproduce this bug?

 Click on field.

### Additional information:

2 thoughts.

1) I have custom pause macros for planed mid print pause, lots of beeping etc.  I would love to be able to use those with this function.

2) It would be amazing to have the option to pause the print at infill rather than the beginning of a new layer. To minimize the very likely possibility of a surface artifact.  I use Cura, I know it identifies infill in the Gcode. That would make this amazing.

meteyou, 2023-02-23:
Could you let me know if you updated your mainsail-config? The button for this feature should not be displayed if you don't have the new gcodes for it.

Just check the gcode. There is no possibility to have fancy special pause places or not because the moment of the layer change comes from your slicer. it just interrupts the layer change gcode. nothing more.

zellneralex, 2023-02-23:
First of all did you follow the instructions mentioned here https://github.com/mainsail-crew/mainsail-config#new-feature-pause-at-next-layer-and-pause-at-layer-.

For your 1st question: you should be able to use your own Pause but **you need to 2 macros** of the mainsail.cfg
For your 2nd question: we will be happy to integrate it, if you find a way that Klipper sends a message when infill starts 

And as @meteyou ask do you have the latest mainsail.cfg (could be found in the linked repo) and also did you include it?

PhilBaz, 2023-02-23:
@meteyou 

No, I didn't update my mainsail.cfg.  and I can see the button.  Sorry I didn't realize changes were needed. I clicked on the feature link in the release message but didn't see that I needed to update.

I understand. I though maybe the gcode would also indicate the printing of infill and the pause could be delayed until there. I know Cura creates comments in the gcode when a wall vs skin vs infill is printing.  But, of course don't know how to code or the limits of what can be implemented.

zellneralex, 2023-02-23:
@PhilBaz that you see the button is a bug. Follow the link I posted there you see the needed slicer changes.

The problem is that cura does comment and PrusaSlicer does other comments and … does other comment and so one.
What we did was enhancing a feature that we added to get a more accurate layer numbering.

PhilBaz, 2023-02-23:
@zellneralex 

No I didn't. How was I supposed to know about those instructions? 

If they were mentioned here
[image]

Or here
[image]

I would have.

For your 1st question: you should be able to use your own Pause but you need the 3 macros of the mainsail.cfg

Awesome, thanks.

For your 2nd question: we will be happy to integrate it, if you find a way that Klipper sends a message when infill starts

I thought this might be a possibility as Cura creates comments in the gcode when a wall vs skin vs infill is printing. I don't know how to code or the limits of what can be implemented. But this would be amazing. If its not possible I wouldn't know that. I just like 3d printers, and its a good idea.

zellneralex, 2023-02-23:
For cura you could also use https://github.com/pedrolamas/klipper-preprocessor to get the macros in. As said cura is one of many …

zellneralex, 2023-02-23:
That is a beta release we need to get a better documentation for the release.

PhilBaz, 2023-02-23:
@zellneralex 

> That is a beta release we need to get a better documentation for the release.

No problem. I just had no clue to do it. 

Hey, at least a genuine bug was discovered in a round about way.

PhilBaz, 2023-02-24:
@zellneralex  @meteyou 

> For your 1st question: you should be able to use your own Pause but **you need the 3 macros** of the mainsail.cfg For your 2nd question: we will be happy to integrate it, if you find a way that Klipper sends a message when infill starts

Hi Alex,

I mentioned this idea to Pedro the developer of the klipper preprocessor for Cura.  He suggested how it might work. I think he proposes what you are requesting above.

https://github.com/pedrolamas/klipper-preprocessor/issues/1 

I'm not a programmer, just a keen user of 3d software and 3d printers. Is there anything I can do to advance this idea? I guess someone on the klipper side would need to create the new parameter that Pedro purposes.

zellneralex, 2023-02-24:
Yes he suggested a change that someone needs to do. Any programmer can do that, the original implementation on the Klipper side is from me. The hard part will be to Konvente Kevin (The main Klipper developer) to merge that

But honestly why you want the pause at infill, I see that pause for color changes or the insertion of nuts or magnets and all that will work at a layer start. So what is the reason to pause at infill?

PhilBaz, 2023-02-24:
@zellneralex 

To prevent surface artifacts. Pause always has some small imperfection with flow, due to the retract, de-retract, and some ooze.

If pause comes back to an outer wall you will see a flaw in that spot. If it comes back to infill there will be no flaw.

I will mainly use pause to swap out rolls that are running out. I print a lot so this is a regular need. For me I think there is real value in this idea as it will eliminate surface flaws that often result from a pause for a filament swap.

zellneralex, 2023-02-24:
I see, as said the hard part will be to convince Kevin. And we would also need to enhance the moonraker post processor to get that feature available for more slicer .

PhilBaz, 2023-02-24:
@zellneralex 

How can I help?

PhilBaz, 2023-03-01:
@zellneralex 

Hi Alex,

I made a post on the Klipper forum in hopes that Kevin O'Conner will consider implementing the parameter Pedro proposed that would allow for pause at infill.

Here is the post. 

[klipper.discourse.group/t/fr-new-parameter-on-set-print-stats-for-section-type-infill-wall-skin-etc/6943](https://klipper.discourse.group/t/fr-new-parameter-on-set-print-stats-for-section-type-infill-wall-skin-etc/6943)

Ill send kevin a message through the form asking him to look at the post. 

I do think having Klipper be aware of what type of feature it is printing may have additional benefits beyond delaying a Pause until infill. 

Anyhow, I hope I did not overstate your willingness to entertain the idea of implementing this into Mainsail.

zellneralex, 2023-03-01:
@PhilBaz we will see, as said cura is only a small part of it. To call it a feature it need to be solvable at the common slicer. I would say the same that support exclude object (SuperSlicer, PrusaSlicer, Slic3r, Cura, IdeaMaker)  would be a good start.
And that’s the hard part here, analyzing there code output see if the add there Type as comment and than do a preprocessor for them or better integrate that all in the prepprocessor used by moonraker for exclude object.
So you see what I mean you may be only interested at a solution insight cura but to get the thing „flying“ we need to look at the bigger picture. 
I personally have no interest in investing my spare time in that right know. There is more important stuff to solve on my plate atm. 
Using that then in the macros of the 2 Pause functions should not be a big deal at all if we get it in.

But we need to do that after V2.5 as an FR as the real bug is that you should not have seen the feature at all as your Goode does not uses the needed macros yet.

PhilBaz, 2023-03-01:
@zellneralex 

Hi Alex,

I hear exactly what you are saying. Its a slicer issue too. 

I'm happy to help with the research and analyze/characterize the outputs from the different slicers. If and when there is a possibility of having development of this functionality go forward.

I'm a researcher and I'm good with a spreadsheet.

Also, Pedrolamas said he would integrate the functionality into his preprocessor. I don't know if that saves any work. Maybe his work will also be suitable for the moonraker preprocessor?

I guess we will see if Kevin is receptive first.

Anyhow, I hope I'm not misreading your Intention.  I'm hearing.... you might be willing to entertain the development of this feature in the future but, a) You don't want to be responsible for all the footwork on slicers and moonraker b) The parameter must be implemented in Klipper for this to work.    Is that right?

PR #1291 (2023-03-10): fix: only display PAUSE AT LAYER button, when the macros exists
fixes #1283
closed 2023-03-10 by meteyou