mainsail-crew/mainsail #1286: Documentation gap: webcam & manual installation (mjpg_streamer webcam)

gcoppex, 2023-02-25:
### Mainsail Version:

2.4.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

When installing Mainsail using the ["Manual setup"](https://docs.mainsail.xyz/setup/manual-setup) on a Raspian 32 bits, there is no installation of `mjpg_streamer`. Hence the webcam support guide won't work.

(Side note, `mjpeg-streamer` documentation is slightly outdated and `libjpeg9-dev` needs to be installed instead of `libjpeg8-dev`)

### What did you expect to happen instead?

- Add a section in the ["Manual setup"](https://docs.mainsail.xyz/setup/manual-setup)  about how to (compile and) install `mjpg_streamer`.
- Add a note in the [Webcam guide page](https://docs.mainsail.xyz/features/webcams) that links to the aforementioned new section of the ["Manual setup"](https://docs.mainsail.xyz/setup/manual-setup)

### How to reproduce this bug?

Manually install Mainsail, using the official ["Manual setup"](https://docs.mainsail.xyz/setup/manual-setup)  and then try to activate the webcam using the [Webcam guide page](https://docs.mainsail.xyz/features/webcams)

gcoppex, 2023-02-26:
Another remark along the way: the gcode path for the Virtual SD Card in the [Required configuration](https://docs.mainsail.xyz/configuration#virtual-sd-card) seem to be outdated. 

I experienced a "File not found" error, when printing a gcode.

According to [rad126](https://www.reddit.com/r/klippers/comments/nsj881/comment/iw579gn/?utm_source=reddit&utm_medium=web2x&context=3): _the path should be replaced with the klipper instance name for each printer_.

Now my `printer.cfg` reads:
```
[virtual_sdcard]
path: /home/pi/printer_data/gcodes
```

meteyou, 2023-02-26:
Mainsail Docs have a own repo for issues, but we are working on new docs, so all of them is currently outdated.

closed 2023-02-26 by meteyou

gcoppex, 2023-02-26:
Ok, opened an issue there: https://github.com/mainsail-crew/docs/issues/37

meteyou, 2023-02-26:
As I wrote above... We are working on a new docs solution. So also the docs repo will be closed soon.