mainsail-crew/mainsail #1287: Docker image not available?

mikerocklewitz, 2023-02-25:
### Mainsail Version:

Current

### Browser:

Chrome

### Device:

Other (Please tell us below)

### Operating System:

Other (Please tell us below)

### What happened?

I'm running mainsail on my unraid server as a docker container.

I dont know exactly when, but at some point in the last week or two I am no longer able to get updates from the repo.  It says not available.  I have tried changing the repo from main to edge and I still get the same "not available" message as shown in this screenshot.
[image]

### What did you expect to happen instead?

When i click check for updates in unraid, it should say up to date or update available.....not "not available"

### How to reproduce this bug?

in unraid try updating the container

meteyou, 2023-02-26:
all images are available: https://github.com/mainsail-crew/mainsail/pkgs/container/mainsail

so it looks like the issue is on your side.

closed 2023-02-26 by meteyou

mikerocklewitz, 2023-03-01:
If anyone else stumbles across this problem, it's an issue with unraid 6.11.5. 

There is a dockerpatch plugin available in community apps to repair it.  More info can be found here: https://forums.unraid.net/topic/134016-fix-to-a-small-issue-with-updating-oci-docker-images/