mainsail-crew/mainsail #1295: Mainsail can't read my C922 pro stream webcam, nor C615

DuckersMcQuack, 2023-03-06:
### Mainsail Version:

2.3.1

### Browser:

Chrome

### Device:

Other (Please tell us below)

### Operating System:

Linux

### What happened?

I can't add the webcam, no image appears with any of the video processers on C922 pro nor C615

### What did you expect to happen instead?

A video feed

### How to reproduce this bug?

Use the mentioned cameras

meteyou, 2023-03-06:
pls check your streamer software. Mainsail only display streams. nothing more. If it cannot display it, there is no stream.

closed 2023-03-06 by meteyou

DuckersMcQuack, 2023-03-07:
> pls check your streamer software. Mainsail only display streams. nothing more. If it cannot display it, there is no stream.

It's not streamed, It simply won't display any image from either of the cam's in the webUI.

meteyou, 2023-03-07:
as i wrote above. pls check your streamer!

DuckersMcQuack, 2023-03-07:
> as i wrote above. pls check your streamer!

Oh sorry, i have a weird thing were i only read every second word. How do i check the streamer? Any config files i have to edit? If so, which ones? As just selecting different streamers and rebooting does nothing sadly.

DuckersMcQuack, 2023-03-07:
I got it working, had to install crowsnest.