mainsail-crew/mainsail #1296: Fun printer travel log to see current print/past print travel distance/filament meters/weight used

DuckersMcQuack, 2023-03-06:
### Requested feature:

I want to request a feature where mainsail logs how long distance in meters/kilometers the printer has traveled for current/past prints, as well as total travel overall since the beginning of mainsail install that we can add to mainsail as a tab/window on the dashboard/next to previously successful finished prints.

### Solves the following problem:

No problem, just some fun statistics

meteyou, 2023-03-06:
mainsail is only a web interface. so there is no possibility for logging anything, when the browser is not opened. so this have to be integrated in Moonraker. All other informations in the history tab are also logged by Moonraker. Mainsail only display it.

Feel free to create a new FR, when these values are integrated in Moonraker. Then we can implement it in Mainsail.

DuckersMcQuack, 2023-03-07:
What's FR?

meteyou, 2023-03-07:
FR = Feature Request. The same what you did here.