mainsail-crew/mainsail #1297: Theme Switch

Steven2k6, 2023-03-06:
### Requested feature:

A switch to change the Theme from dark to Light and backwards.

I made a Vuetify3 Project as example.

[image]

Maybe you can also add a setting to set the colors for the Dark and Light Theme.
I uploaded a zip with the Vue-File.
[Switch-Test.zip](https://github.com/mainsail-crew/mainsail/files/10900195/Switch-Test.zip)

meteyou, 2023-03-06:
Feel free to made a PR, but there are some parts, which also needs some other css settings, because only switching to light mode doesn't fit.

Steven2k6, 2023-03-06:
I don't know if its possible, because I'm relativ new to programming, but isn't it possible to create a public variable for the colors which you can edit in an editor or setting menu?

meteyou, 2023-03-06:
All interface settings are the same function like this. It's only more work, than just a switch. You have to restyle all charts and other functions for the light mode.

Rigattoni, 2023-06-23:
Are there plans and may be an ETA for the light mode?
The dark mode is nice, but sometimes you really need some better contrast as it is given with the dark mode, at least if you are working outside and have a sunny day. Then it is quite hard to see the stuff. 
Mostly the editor is not readable anymore. Yes, I can download the file to edit it in notepad ++... Easier to handle would be to have a white background and contrasting colors in the editor itself.

meteyou, 2023-06-24:
@Rigattoni this is just a hobby project. so we don't have any ETA...

light mode is planned, but there are much more important things i have to implement, before i can implement a light mode. I hoped, that a contributor will fix this missing issue before...

meteyou, 2023-06-24:
I added all missing tags and milestones. Maybe it is better visible now for contributors.

PR #1580 (2023-12-16): feat: light mode ui
## Description

Added a light mode with accompanying option under UI Settings, relying on vuetify defaults (removing explicit "dark" attributes in code etc.) and a theme mixin used to close the gap/centralize to extend consistently to chart options and other specific places where needed

## Related Tickets & Documents

Fixes #1297

## Mobile & Desktop Screenshots/Recordings

[image]
[image]
[image]
closed 2023-12-16 by meteyou