mainsail-crew/mainsail #1298: Print history running total of filament used for each filament roll

asm308, 2023-03-07:
### Requested feature:

Would it be possible to add a column in the history which kept track of the total length of filament used for each different roll of filament you might have.  Would make it easier to know approximately how much was left and be able to better use roll ends more easily.
I guess it would require some way (perhaps from slicer data) to identify the roll you have loaded 

### Solves the following problem:

would help avoid running out of filament mid print

coinstore, 2023-03-10:
Probably similar to https://github.com/mainsail-crew/mainsail/issues/382

meteyou, 2023-03-12:
i see no difference between this FR and the "old" one...

closed 2023-03-12 by meteyou