mainsail-crew/mainsail #1305: Reset Print Statisctics

Steven2k6, 2023-03-14:
### Requested feature:

Because I didn't find the option for it, I write this feature request.
[image]
Recently I got new filament, because my old ran out. Than I went to the Statistics tab to reset the statistics it, because I roughly want to know, how much of my new filament I already used. So maybe you could add an option to reset the statistics.

### Solves the following problem:

Adding a reset button for the statistics.

meteyou, 2023-03-14:
Its already implemented. Goto interface settings > reset > history stats.

closed 2023-03-14 by meteyou

Steven2k6, 2023-03-15:
> Its already implemented. Goto interface settings > reset > history stats.

Where do I find the interface settings?

alexdaums, 2023-05-02:
Hello,

I can't find the function : settings > reset > history stats.
Is there any other way to do it?

Thank you.