mainsail-crew/mainsail #1306: Implement Webcam Scaling options

Drachenkaetzchen, 2023-03-16:
### Requested feature:

Implement customizable scaling of the dedicated "Webcam" view:

- Scale to browser width
- Scale to browser height
- 1:1

### Solves the following problem:

In Mainsail v.2.4, the webcam was scaled to the width of the browser, which allowed the user to somewhat make use of hi-res webcams. 

In Mainsail v2.5, the webcam is scaled to the height of the browser, which makes the image way smaller. 

Mainsail v2.4: 
[image]

Mainsail v2.5:
[image]

Dogfroster, 2023-03-21:
The new view in the webcam tab is useless.
It's way too small especially on a laptop, but even on a 32" monitor it's too small.
UHD camera and yet only mouse cinema.

Phil1988, 2023-11-08:
For me the field of view looks way bigger on 2.5