mainsail-crew/mainsail #1313: WebRTC Webcam Stuck On Connecting

jph290, 2023-03-18:
### Mainsail Version:

2.5.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When I enter the URL used in my browser to view my Web RTC stream that is hosed on the same RPi as MainSail, it just says connecting and never shows the stream.  The url is in the format http://192.168.50.27:8889/3d-printer/.  When used in the address bar of chrome it works just fine.  It also works on the page of the software hosting the WebRTC stream.  I am using this project setup in docker to connect to my wyze cam pan v2, which can not be flashed with the custom wyze firmware.

### What did you expect to happen instead?

The WebRTC video feed to play

### How to reproduce this bug?

Add url into the webcam URL Stream and select the Service WebRTC (camera-streamer).

### Additional information:

Here is the info from the running stream in chrome from the chrome://webrtc-internals page.  
[image]

meteyou, 2023-03-19:
this streamer is (as the name said) only for for the webrtc streamer "camera-streamer".

closed 2023-03-19 by meteyou

jph290, 2023-03-19:
Then the documentation should state that camera-streamer is a specific piece of software.  This is a very generic name for a piece of software.  WebRTC is made for "camera streaming".

meteyou, 2023-03-19:
> Then the documentation should state that camera-streamer is a specific piece of software. This is a very generic name for a piece of software. WebRTC is made for "camera streaming".

Sry, but the name of the streamer is "camera-streamer". This is not our name...