mainsail-crew/mainsail #1315: Different between estimate time from slicer and mainsail

JackyHe398, 2023-03-20:
### Requested feature:

[image]
From the photo, when (the actual time is >  the estimate time from the slicer), it shows "--" only. 

I think that maybe better if it can show how much time it has overrun. Users can have a clear concept about the time of the future print. Slicer is not always accurate. 

### Solves the following problem:

This is nothing related to bugs

meteyou, 2023-03-20:
then i would prefer to check the history, because you don't know how long this print will need in the current state

JackyHe398, 2023-03-21:
Interesting. 

I would like to ask is there any reason not to show that overrun time? Maybe from other perspective, it may not always be a good idea?

meteyou, 2023-03-21:
just my 2 cents. i personally don't see any advantage in displaying them. but if it's wanted by many, i see no reason not to add this feature.

mikebcbc, 2023-05-15:
@meteyou this issue can be closed - it's merged in

meteyou, 2023-05-15:
Thx for the ping

closed 2023-05-15 by meteyou