mainsail-crew/mainsail #1320: References to klipper Configuration Reference

JustKilian, 2023-03-23:
### Requested feature:

Have a link to the klipper docs in the config for each block of config, like in fluidd.

### Solves the following problem:

No problem in particular, but a nice feature to have.

kyledavis417, 2023-03-23:
This would be nice to have, with also having a toggle to opt out of the links for expert users.

JustKilian, 2023-03-23:
> also having a toggle to opt out of the links for expert users.

Yes, that's a nice addition.

oernster, 2023-03-23:
This sounds like a lovely addition.  Yes please.

meteyou, 2023-03-23:
i don't know if this is possible in our editor. mainsail uses codemirror. fluidd uses monaco

JustKilian, 2023-04-04:
> i don't know if this is possible in our editor. mainsail uses codemirror. fluidd uses monaco

Are you and your team able to find out if this is at all possible? Would be a really nice feature. @meteyou

meteyou, 2023-04-04:
@JustKilian sry. i haven't time to check it until now.

forgodtosave, 2023-04-25:
A solution could be to create a codemirror widget for every block of config a bit like "[Boolean Toggle Widgets](https://codemirror.net/examples/decoration/)".
Only with a block widget (which lays between the rows of code so also above each block of config) containing the link to the Klipper config reference.
I started working on it, but I came across one problem. There is no searchable syntax tree to efficiently find every block of config. This tree could be created by defining a new language for the Klipper config. I would like to try this and opened a [FR] (#1355) for this.

meteyou, 2026-09-25:
this was implemented in #1943 and #2537

closed 2026-09-25 by meteyou