mainsail-crew/mainsail #1323: Exclude Object button gone

srbjessen, 2023-03-26:
### Mainsail Version:

2.5.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Exclude object button seems to have been replaced with "Pause on layer" button
[image]
[image]

### What did you expect to happen instead?

When print starts, "Exclude Object" button used to appears next to "Pause" and "Cancel Print"

### How to reproduce this bug?

Happens on every print

closed 2023-03-26 by github-actions[bot]

meteyou, 2023-03-26:
It's not a bug, it's just a wrong config.

srbjessen, 2023-03-26:
Exclude object works, it's in the status panel above (currently missing) M117 messages.
[image]
I just can't see the overview of the buildplate and select an object to cancel, only cancel currently active part.
But i will doublecheck config and ask discord if further issues.

meteyou, 2023-03-26:
Is this print only with one object?

srbjessen, 2023-03-26:
Current print is only one object yes, but it happens with multiple objects too. Will try to reboot/reset everything and double check my configs, might have done something screwy that I've forgotten about.  :-)

meteyou, 2023-03-26:
I hide the button at the top when there is only 1 object, so you can click the layer pause button directly without the drop-down menu.

srbjessen, 2023-03-26:
Is that new in 2.5? That might be what's messing with me. It used to be there, with both 1 and multiples.
It's working now at least. Thanks and sorry :-)

meteyou, 2023-03-26:
yes. it comes with the "pause on layer" feature. if you have both options available, you will see a submenu with both buttons. i don't to have a submenu when there is only 1 object, because object canceling with 1 object is not very useful. So i hide this option to have only the "pause on layer" function in the toolbar.

srbjessen, 2023-03-26:
Awesome, then it's once again just me :-)
Keep up the good work. 👍