mainsail-crew/mainsail #1324: SCREWS_TILT_CALCULATE dialog empty when re-running SCREWS_TILT_CALCULATE without FIRMWARE_RESTART

sjorge, 2023-03-26:
### Mainsail Version:

v2.5.0 

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

macOS

### What happened?

First time running `SCREWS_TILT_CALCULATE` the nice info dialog with how to rotate the screws pops up. After ajusting and rerunning `SCREWS_TILT_CALCULATE` the dialog pops up but is now empty. The console stills shows the correct information though.

### What did you expect to happen instead?

The dialog to also have the info the 2nd time running `SCREWS_TILT_CALCULATE`.

### How to reproduce this bug?

- home (`G28`)
- `SCREWS_TILT_CALCULATE`
- <adjust as suggested by the dialog>
- `SCREWS_TILT_CALCULATE`
- <dialog is now empty>

### Additional information:

My klipper config snipper
```
[screws_tilt_adjust]
screw1: -6, 28
screw1_name: front left screw
screw2: 158, 28
screw2_name: front right screw
screw3: 158, 193
screw3_name: rear right screw
screw4: -6, 193
screw4_name: rear left screw
horizontal_move_z: 10
speed: 50
screw_thread: CCW-M4
```

meteyou, 2023-03-26:
pls post a screenshot from your browser console

sjorge, 2023-03-26:
first time
[image]

second time
[image]

console log:
```
16:14:12
rear left screw : x=-6.0, y=193.0, z=-0.02083 : adjust CCW 00:01
16:14:12
rear right screw : x=158.0, y=193.0, z=-0.06125 : adjust CCW 00:04
16:14:12
front right screw : x=158.0, y=28.0, z=-0.08167 : adjust CCW 00:06
16:14:12
front left screw (base) : x=-6.0, y=28.0, z=-0.01042
16:14:12
01:20 means 1 full turn and 20 minutes, CW=clockwise, CCW=counter-clockwise
16:14:12
probe at -6.000,193.000 is z=-0.022500
16:14:11
probe at -6.000,193.000 is z=-0.022500
16:14:10
probe at -6.000,193.000 is z=-0.022500
16:14:10
probe at -6.000,193.000 is z=-0.022500
16:14:09
probe at -6.000,193.000 is z=-0.020000
16:14:08
probe at -6.000,193.000 is z=-0.015000
16:13:59
probe at 158.000,193.000 is z=-0.062500
16:13:58
probe at 158.000,193.000 is z=-0.062500
16:13:58
probe at 158.000,193.000 is z=-0.062500
16:13:57
probe at 158.000,193.000 is z=-0.062500
16:13:56
probe at 158.000,193.000 is z=-0.060000
16:13:56
probe at 158.000,193.000 is z=-0.057500
16:13:46
probe at 158.000,28.000 is z=-0.080000
16:13:46
probe at 158.000,28.000 is z=-0.080000
16:13:45
probe at 158.000,28.000 is z=-0.082500
16:13:44
probe at 158.000,28.000 is z=-0.082500
16:13:44
probe at 158.000,28.000 is z=-0.082500
16:13:43
probe at 158.000,28.000 is z=-0.082500
16:13:34
probe at -6.000,28.000 is z=-0.012500
16:13:33
probe at -6.000,28.000 is z=-0.012500
16:13:32
probe at -6.000,28.000 is z=-0.012500
16:13:32
probe at -6.000,28.000 is z=-0.010000
16:13:31
probe at -6.000,28.000 is z=-0.010000
16:13:30
probe at -6.000,28.000 is z=-0.005000
16:13:24
SCREWS_TILT_CALCULATE
16:13:00
rear left screw : x=-6.0, y=193.0, z=-0.03042 : adjust CCW 00:00
16:13:00
rear right screw : x=158.0, y=193.0, z=-0.07458 : adjust CCW 00:04
16:13:00
front right screw : x=158.0, y=28.0, z=-0.10167 : adjust CCW 00:06
16:13:00
front left screw (base) : x=-6.0, y=28.0, z=-0.02625
16:13:00
01:20 means 1 full turn and 20 minutes, CW=clockwise, CCW=counter-clockwise
16:13:00
probe at -6.000,193.000 is z=-0.032500
16:12:59
probe at -6.000,193.000 is z=-0.032500
16:12:58
probe at -6.000,193.000 is z=-0.032500
16:12:57
probe at -6.000,193.000 is z=-0.030000
16:12:57
probe at -6.000,193.000 is z=-0.030000
16:12:56
probe at -6.000,193.000 is z=-0.025000
16:12:47
probe at 158.000,193.000 is z=-0.075000
16:12:46
probe at 158.000,193.000 is z=-0.075000
16:12:46
probe at 158.000,193.000 is z=-0.075000
16:12:45
probe at 158.000,193.000 is z=-0.075000
16:12:44
probe at 158.000,193.000 is z=-0.075000
16:12:43
probe at 158.000,193.000 is z=-0.072500
16:12:34
probe at 158.000,28.000 is z=-0.097500
16:12:34
probe at 158.000,28.000 is z=-0.100000
16:12:33
probe at 158.000,28.000 is z=-0.100000
16:12:32
probe at 158.000,28.000 is z=-0.102500
16:12:31
probe at 158.000,28.000 is z=-0.105000
16:12:31
probe at 158.000,28.000 is z=-0.105000
16:12:22
probe at -6.000,28.000 is z=-0.027500
16:12:21
probe at -6.000,28.000 is z=-0.025000
16:12:20
probe at -6.000,28.000 is z=-0.027500
16:12:19
probe at -6.000,28.000 is z=-0.027500
16:12:19
probe at -6.000,28.000 is z=-0.027500
16:12:18
probe at -6.000,28.000 is z=-0.022500
16:12:11
SCREWS_TILT_CALCULATE
16:11:33
G28
```

browser console:
[image]

geonizeli, 2023-03-31:
I am having this issue too. Sometimes I have the page opened on multiple devices and when triggering from one, the result is only showed on others.

jmwhelan, 2023-03-31:
is there a way to turn off that dialog? I prefer to just read the console...

meteyou, 2023-03-31:
i found this issue and fixed it in the PR #1329

meteyou, 2023-03-31:
@jmwhelan will come with the PR #1319

jmwhelan, 2023-03-31:
> @jmwhelan will come with the PR #1319

Love you! Lol

PR #1329 (2023-04-01): fix: fix issue of empty Screws tilt adjust helper dialog
fixes: #1324