mainsail-crew/mainsail #1331: Hard to see what is selected in dark mode text editor

henrikba, 2023-04-03:
### Mainsail Version:

2.5.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When editing config files, I often use keyboard shortcuts to select text.
The editor shows dark grey background, slightly lighter grey to show which line is "active" (eg where my cursor is), and a very slightly lighter grey background to indicate selected text.

[image]

The image shows "waiting" is the selected word. In 1:1 pixels/my screen setup, it is really hard to see cursor and selection.

### What did you expect to happen instead?

Larger contrast between selected word and active line

### How to reproduce this bug?

Select text in editor

henrikba, 2023-04-03:
FWIW; I did play around with [codemirror](https://codemirror.net/) (the syntax highlighter that apparently is used). 
By default, this has a lot higher contrast between selected text and "active line":
[image]

Otann, 2023-05-13:
Google had lead me here on the topic of how to change CodeMirror theme.

meteyou, 2023-05-15:
@Otann you can only change it in the sourcecode itself... there is no option in the GUI for this.