mainsail-crew/mainsail #1348: Add Multi-theme Support And JS Theme Support

QuinnDamerell, 2023-04-23:
### Requested feature:

I'm coming at this from the perspective of a plugin developer. It would be fantastic if there were some way of extending the UI, even just adding a small panel for the plugin config or settings. I'm sure adding full HTML injection support would be much harder, so I was thinking, as a minium, if Mainsail supported allowing themes to inject a custom JS file, and for Mainsail to support multiple themes at the same time, it would do the trick. With a custom JS file, plugin devs could (in a hacky way) add a little UI, call APIs, or whatever they need. minimum

### Solves the following problem:

This would allow users and plugin devs to do more powerful things!

meteyou, 2023-04-23:
but plugins are very useless without moonraker or klipper support. thats why we dont want to implement it.

QuinnDamerell, 2023-04-23:
I don't know if that's necessarily true; think of all the things users did with plugins like [Greasemonkey](https://addons.mozilla.org/en-US/firefox/addon/greasemonkey/) before more official browser extensions came out. I'm sure there are at least a few users out there who have good ideas that could augment Mainsail to work better for them.

In my case, I have a full plug-in running on the device, so I was looking for a way to add anything to the main UI for it. I found the theme stuff but then saw there can only be one theme active at a time, and also, there's no way to add UI or JS. That led me down the thought path of how Mainsail could add support to allow plugins running on the device to show some UI.

My plug-in requires linking with a service, which has to be done via the command line when installed. It's not ideal for users, and some users skip it and don't realize it. So having some way to show UI in Mainsail would be amazing. Using just a JS file is a little janky and can be a little fragile, but I feel like as a stop-gap it would be a rather easy way for Mainsail to support some kind of plug-in UI extensions.

meteyou, 2023-04-23:
How do you want to load multiple themes at the same time? Which background image should be used for example? Themes are based on naming conventions to change default files.

QuinnDamerell, 2023-04-23:
Yeah, that’s fair. I was thinking to extend themes because it’s a concept that already exists. But you’re right I doesn’t seems to fit well. 

Maybe like the .theme folder, there could be a .extensions folder that would contain .js files, and any js file put in the dir would be included into the webpage?

meteyou, 2023-04-24:
I only see more negative things with "just loading random js" files. There are no hooks or something, that you can really work with these "extensions".

QuinnDamerell, 2023-04-26:
Yeah, that's fair. I'm sure they might be powerful to some but would cause craziness for others. :D 

In general, it would be great if installed system plugins could integrate into Mainsail somehow to show some UI. I know that's a tall order, but it would be amazing if it ever became a thing!

closed 2023-04-26 by QuinnDamerell