mainsail-crew/mainsail #1355: [FR] Codemirror Language Package for printer.cfg

forgodtosave, 2023-04-25:
### Requested feature:

A Codemirror Language Package for printer.cfg. It would enable a lot of useful features, like 
- error highlighting without "Restart&Save"  
- better code highlighting  
- auto indentation ... 

It would also create a syntax tree, which would be useful to find blocks of config to resolve #1320

### Solves the following problem:

- No static error checking in the config file 
- no auto indentation 
- flat syntax tree 

### Additional information:

I would like to try it and would start based on this example, but I’m not sure if there will be any license problems or what a good directory structure would look like.

meteyou, 2023-04-25:
pls check the open PRs. most of these things are already did...

forgodtosave, 2023-04-26:
Sorry but I can't find anything regarding this in the open PRs. There is only text highlighting for gcode but nothing regarding the printer config and a grammar for it to create a language package. Could you maybe send a link?

meteyou, 2023-04-30:
@forgodtosave ohhh that sounds really nice!

can you use this PR/branch to check the syntax highlighting for your language package -> https://github.com/mainsail-crew/mainsail/pull/1200

forgodtosave, 2023-05-03:
Sorry, but I don't think so, at least in the current state. Lezer supports [grammar nesting](https://lezer.codemirror.net/docs/guide/#grammar-nesting) and it should even be possible between a lezer parser and a stream parser. The only problem is that in #1200 the streamParserJinja2 has been removed and integrated into streamParserKlipperConfig. Of course it would be possible to nest the new parser of #1200 into the new language, but it would be quite ugly because it would be a klipper/jinja2 paser in a klipper parser. 

Another question: why did you implement a custom jinja2 streamParser? There is a [public version](https://github.com/codemirror/legacy-modes#user-content-jinja2) available.

meteyou, 2023-05-04:
This was meant as a hint that you should follow this syntax highlighting instead of the current one from Mainsail.

I have not implemented this. This was a contribution...

zellneralex, 2023-05-04:
@forgodtosave we tested the public parser but Klipper has e.g. modified start and end variable strings. Testing with my config (https://github.com/zellneralex/klipper_config) showed in many areas that the default parser failed.

BTW our custom parser is also not 100% perfect but the best we have so far.

forgodtosave, 2023-05-09:
@zellneralex ok then of course it makes sense to use the custom parser.

@meteyou the new streamparser definitely looks better than the current one. However, I think it would be great if there were more than just syntax highlighting. I started with a grammar for the KlipperConfig and that seems to work quite well. Everything related to Jinja2, is currently outsourced to the old parser. Of course, that could also be the new one from #1200, but then everything related to the KlipperConfig in it would be unnecessary.

meteyou, 2023-05-09:
@forgodtosave I would also love a complete language package. One feature that I personally always miss is a collapse section function. I think this is also only available in language packages.