mainsail-crew/mainsail #1356: Cura overlap mesh does not show in cancel dialogue

PhilBaz, 2023-04-26:
### Mainsail Version:

current stable

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

https://github.com/kageurufu/preprocess_cancellation/issues/22

I'm using object cancel preprocessing through moonraker, not the Cura script.

I have a mesh where upper layers are modified by an overlapping mesh. In Cura these overlapping meshes are not to be printed but are used to modify the settings for certain portion of the mesh. In Cura we can use 'support blockers' as this mesh. It is the fastest so that's what I use. A support blocker receives the name 'Eraser', with no type designation (.stl .3mf etc.)

This 'Eraser' mesh is not visible in the mainsail object cancel dialogue or available in the object list on the side. I am able to cancel the lower part of the mesh before the overlapping portion, but the upper part remains and of course ruins the print.

The preprocessor sees and labels the 'Eraser' mesh with no file type. If I replace "Eraser' with 'Eraser_3mf' I can then see the mesh in mainsails cancel dialogue. ***I'm not sure if I can visually see it and distinguish it from the mesh below it, but it is labelled in the mesh list.

Not sure exactly where to report this so ill cross post at mainsail to.

Thanks for the tools!

[image]

### What did you expect to happen instead?

One of two things...

1) When I cancel the underlying object (of which some portion is modified by the overlap) the overlap section is also cancelled.

2) I am able to visually distinguish the overlapping section in the object cancel dialogue as well as see it listed on the side. 

### How to reproduce this bug?

described above

### Additional information:

Thanks again for all you do!

meteyou, 2023-04-26:
Mainsail cannot read/see any gcode. It just display the Klipper current state.

closed 2023-04-26 by meteyou

PhilBaz, 2023-04-26:
Hi Stefan, 

This issue is only about object cancelation dialog already in mainsail. There is a problem. Maybe I didn't explain it clearly?

PhilBaz, 2023-04-28:
@meteyou Hi Stefan, please see above. Maybe I didn't explain clearly earlier. Its not a feature but a bug.