mainsail-crew/mainsail #1358: Printers either disconnect or display as disconnected

Darkmann12, 2023-04-29:
### Mainsail Version:

2.5.1

### Browser:

Other (Please tell us below)

### Device:

Laptop

### Operating System:

Windows

### What happened?

Browser is Brave. 

https://user-images.githubusercontent.com/83469818/235286232-e2906cff-aab2-4bf9-abe6-8c5dfc332714.mp4

https://user-images.githubusercontent.com/83469818/235286233-cb567c21-c577-4412-9762-b38af3d224c1.mp4

Upon switching to a different printer, either via the dropdown or the "printers" main UI tab, the previous printer immediately appears to disconnect and requires hitting "reconnect" in the "printers" main UI tab before switching is functional again. In the "printers" tab of the settings popup, it is visible that the printer shows unreachable/disconnected but immediately "connects" as soon as the details are updated (with no changes).

[mr.log](https://github.com/mainsail-crew/mainsail/files/11358479/mr.log) 

### What did you expect to happen instead?

I expected both printers to be switchable from and to at immediate notice, displaying "standby" in the dropdown instead of "disconnected".

### How to reproduce this bug?

Simply trying to switch between my printers, either via dropdown or tab. I don't know exactly what caused this issue since I have never seen it in a fully working state.

Darkmann12, 2023-04-29:
Additional information I didn't think about!
- Sometimes it works for a tiny amount of time, I can switch maybe 1 or 2 times before it shows "disconnected".
- Both printers are connected to the same host as is visible in one of the videos - on different ports.

meteyou, 2023-06-18:
Sry... I cannot reproduce this issue with Google Chrome & Safari.

meteyou, 2023-07-26:
closed repo, because of no answer and no reproduce....

closed 2023-07-26 by meteyou