mainsail-crew/mainsail #1362: Object cancel dialogue doesn't recognize objects.

PhilBaz, 2023-05-01:
### Mainsail Version:

current 

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Object identified in pre-processor doesn't appear in object cancel dialogue. The objects name is 'Eraser'. 

[image]

### What did you expect to happen instead?

'Eraser' is a printed object and should be shown in the object cancel dialogue. Visually on the display and in the list.

### How to reproduce this bug?

Slice a file in Cura and use a mesh to 'modify settings for overlap'. As shown here: https://support.makerbot.com/s/article/1667417981430

PhilBaz, 2023-05-02:
@meteyou  Hi Stefan, yes looks like you are right. After doing some testing it looks like it working fine.

 Ill dig into the file where it was not working and see if I can understand why. Its a complex slice in Cura with many per-object settings and overlapping settings. I thought it was mainsail as when i edited
 EXCLUDE_OBJECT_DEFINE NAME=Eraser, to be EXCLUDE_OBJECT_DEFINE NAME=Eraser_3mf, It worked suddenly. 

Sorry about that.

meteyou, 2023-05-02:
pls close this issue, when all works fine in your case...

closed 2023-05-02 by meteyou