mainsail-crew/mainsail #1373: Feature Request: Allow Folder Uploads

AudreyAP, 2023-05-03:
### Requested feature:

Users should be able to drag-and-drop a folder into the Mainsail file upload dialogue. Mainsail currently supports batch file uploads, but not uploads of folders directly. 

### Solves the following problem:

Uploading a folder currently requires a user to create the folder, navigate into it, then upload the files inside as a batch. This needs to be repeated for nested directories, which can be a huge pain. 

### Additional information:

It would be great if this could work both in the g-code file section and printer configuration section.

NiciTheNici, 2023-07-11:
Looks like this is still an issue. Uploading a folder just hangs the entire thing and then throws a "Cannot upload file" error.

This would be great to see in the future! Especially for people (like me) who just installed mainsail and want to copy over their entire old sd card :)

meteyou, 2023-07-12:
yep. all issues/FR with the state "open" are not implemented until now. but every comment to FR helps me to find the most important FR.

meteyou, 2024-11-23:
Moved from Milestone v2.13.0 to v2.14.0, because of some implemention issues with different browsers.