mainsail-crew/mainsail #1379: Allow to start any print from queue not only first one

ristein, 2023-05-06:
### Mainsail Version:

2.5.1

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Having multiple files in queue but I can't use it as a todo-list because I can only start the job added first from queue screen.

### What did you expect to happen instead?

Should be possible to start any file from queue not having to go look for it in files tab

### How to reproduce this bug?

add two file to queue
try to start second one

meteyou, 2023-05-06:
Why should this be a bug?

ristein, 2023-05-06:
Why should this be intentional?

th33xitus, 2023-05-06:
> Why should this be intentional?

It's currently the intended behavior. Should this be rather a feature request instead?

ristein, 2023-05-06:
that would be great

meteyou, 2024-02-03:
fixed with the PR #1692

closed 2024-02-03 by meteyou