mainsail-crew/mainsail #1380: Configuration for default camera

shaunmulligan, 2023-05-07:
### Requested feature:

I have a setup of prind (klipper in docker containers) that I deploy on an Android phone and I then have a MJPEG streaming camera app on the same app. Setting up the camera works perfectly and I can configure it from the UI which is great. However, I would like a way to configure it from the mainsail.conf or something so that it is preconfigured at boot time and doesn't require any additional user input. 

### Solves the following problem:

Improves user experience as they don't need to configure the camera themselves if its a standard setup.

meteyou, 2023-05-07:
Do I understand this question correctly?

Is it about the use case that your docker image for a specific printer at the customer/user has already preconfigured the webcam at the first boot, or does your docker image lose this setting every time it starts?

shaunmulligan, 2023-05-07:
The camera is already setup to go, but the user has to navigate to settings and change the URL in the UI. I would love to be able to define the webcam URL in mainsail.conf. so it just works when the user boots up for the first time.

meteyou, 2023-05-07:
Ok. There exist two variants you can use for this use case.

1. You can use the moonraker.conf. You can add webcams in this config file but cannot edit via Mainsail.
(https://moonraker.readthedocs.io/en/latest/configuration/#webcam)

2. You can use the default.json in the theme directory. With this file, you can change all interface settings in Mainsail, which will be loaded at the first boot (or factory reset of the interface settings).
(Add your webcam in Mainsail, Click on Interface Settings > General > Backup, Copy this file to your config directory in the .theme directory with the name default.json)

shaunmulligan, 2023-05-08:
okay awesome! it seem like #2 is what I am looking for, thanks!!

closed 2023-05-08 by meteyou