mainsail-crew/mainsail #1382: Configure the webcam to automatically use the say address it being served from

shaunmulligan, 2023-05-08:
### Requested feature:

It would be great it we could configure the web camera url to use the same URL the mainsail UI is rendered from. For instance if we are view mainsail on http://raspberrypi.local, then the camera would default to `http://raspberrypi.local:8080/stream` and similarly if you were viewing from a domain like example.com it would rewrite the camera to use `http://example.com:8080/stream`. Is this possible today or would it be easy to add as an option?

### Solves the following problem:

allows the user to view the camera feed both locally and remotely

meteyou, 2023-05-08:
That's the reason for the default redirects. /webcam is redirerected to :8080. So the URL /webcam/stream use the same URL.

closed 2023-05-08 by meteyou

shaunmulligan, 2023-05-08:
hmm, it doesn't seem to work as expected for me, maybe I am missing something? See below I have two camera configurations, when I set the IP it works perfectly, but if I just leave the default with /webcam it just get a black screen.

[image]

[image]

If I understand you correct, leaving the default as /webcam should be equivalent to having the http://IP_address:8080/webcam when mainsail is served from <IP_address> ? 

Is there something I need to configure to have the default redirect work like that?

meteyou, 2023-05-08:
Is 8080 your streamer, or your Webserver?

shaunmulligan, 2023-05-08:
the mainsail is served on 80 and the camera stream is on 8080

meteyou, 2023-05-08:
pls share your nginx config

shaunmulligan, 2023-05-08:
I don’t think I have an ngnix config, I’m using https://github.com/mkuf/prind which I think uses traefik. My setup is unusual as I have Docker compose running in an android app and then the camera stream is served from another app on the same android phone on port 8080

shaunmulligan, 2023-05-10:
Ah okay, so perhaps if I set the traefik labels up it could work? The one thing that is confusing with this approach is that I don't actually have a `webcam` container, the camera is served from a different android app, so traefik doesn't really know about it. Perhaps I can just set those labels on the mainsail container?  Sorry for all the questions, I don't have a great handle on how these reverse proxies work :/

meteyou, 2023-05-10:
@shaunmulligan Sorry. I can't help you with this part. This has nothing to do with Mainsail, and I am not the maintainer of these containers.