mainsail-crew/mainsail #1389: Maintenance Tracking

axseb, 2023-05-12:
### Requested feature:

I'd like to see a maintenance tracker for the printers history. 
You could enter a manual entry to the history descibing what has been done (e.g. changed cabling of your belt, lubing the bearings, etc...) 
That can be used for a statisic like print time since last maintenance as well as adving users to perform maintenance after XXXhours of printing. 

For a specific print history entry I'd like to have the option to change if a print was successfull or not. Also add some kind of failiure description like "bed adhesion", "hardware issue", "intentional fail / User aborted". 
That would help to track reoccuring issues.

RuChef, 2023-05-19:
I don't think this is a good idea for the Print History, but it would be a useful feature in the Machine page as a new Maintenance panel.

meteyou, 2023-05-19:
This needs the print history for Kalkulation and if I don't do it in the print history, the first feature request will be "I will see which prints was between it"... Do that's the reason, why I implement it in print history.

closed 2024-04-28 by meteyou