mainsail-crew/mainsail #1396: [Webcam] Add an option to embed a webcam stream using iframe

scmanjarrez, 2023-05-21:
### Requested feature:

Add a new option in the webcam settings to create an iframe which just loads a URL with the streaming. This shift the cpu load to the web browser instead of the mainsail host.

### Solves the following problem:

I've been tinkering with go2rtc and achieved a no delay stream (http://192.168.1.38:1984/stream.html?src=yicam), however when I try to add the stream to mainsail, the only options are mjpeg or ipcam which add around 5-15s of delay.

scmanjarrez, 2023-05-21:
PoC

https://github.com/mainsail-crew/mainsail/assets/9284124/2054db26-65bd-4cdb-80d4-ed3935606ad1

meteyou, 2023-05-21:
i dont think that iframe is a good integration for that. pls checkout the last PRs about webcam clients.

closed 2023-05-21 by scmanjarrez

scmanjarrez, 2023-05-21:
To whoever interested in a workaround: 
- Create a webcam in mainsail with the urls for stream and snapshot and choose HLS Stream. (This webcam won't load, but we need the node created in the webpage)
- Install a userscript addon in your browser: tampermonkey, greasemonkey, violentmonkey etc
- Create a userscript with the following content:
```userscript
// ==UserScript==
// @name        Mainsail webcam
// @namespace   Violentmonkey Scripts
// @match       http://your_mainsail_machine_ip:7126/
// @grant       none
// @version     1.0
// @author      scmanjarrez
// @description 5/21/2023, 4:40:59 PM
// ==/UserScript==

(function() {
    'use strict';

    // Function to replace the video tag with an iframe
    function replaceVideoWithIframe() {
        var videoElement = document.querySelector('div[data-v-4abc3b36] video.webcamImage');

        if (videoElement) {
            var videoSrc = videoElement.src;
            var iframeElement = document.createElement('iframe');

            iframeElement.src = "http://your_go2rtc_machine_ip:1984/stream.html?src=your_cam_name";
            iframeElement.allow = 'autoplay';
            iframeElement.muted = true;
            iframeElement.style.transform = 'none';
            iframeElement.style.aspectRatio = '1.77778 / 1';
            iframeElement.style.maxHeight = '345px';
            iframeElement.style.width = '100%';
            iframeElement.classList.add('webcamImage');

            videoElement.parentNode.style.height = '300px';

            videoElement.parentNode.replaceChild(iframeElement, videoElement);
        }
    }

    // Wait for the DOM to be loaded
    window.addEventListener('DOMContentLoaded', function() {
        setTimeout(replaceVideoWithIframe, 5000);
    });
})();
```

Also, in your go2rtc.yaml configuration add the following setting to allow CORS policy:
```yaml
api:
  origin: "*"
```

> If you're using docker like me, mount a folder like this: `-v $PWD/config:/config` and the create
the file **go2rtc.yaml** in the folder config.

Don't forget to update `your_mainsail_machine_ip` in userscript `@match` and `your_go2rtc_machine_ip`
in `iframe.src`

> If the iframe is not created, try to increase the timeout `setTimeout(replaceVideoWithIframe, 5000);`. I'm using 5 seconds, but increase it to something like 10.

PoC:

https://github.com/mainsail-crew/mainsail/assets/9284124/309f9e4f-2fc2-4a93-be5f-3ca75e3a307d

vajonam, 2023-11-11:
@scmanjarrez  this works so much better than all the WebRTC options that are in mainsail at the moment. Iframe removes the issues with CORS which is a pain to setup when dealing with reverse proxies.. so this is great!

PR #1651 (2023-12-10): feat(webcam): add support for go2rtc webrtc
## Description

This PR add support for go2rtc webrtc streams.

## Related Tickets & Documents

fixes #1396

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

- add this streamer to the docs