mainsail-crew/mainsail #1397: Power down USB port when printer not in use

hughesjs, 2023-05-26:
### Requested feature:

Assuming we're using an RPi with MainSail OS, a lot of printers will draw 5V power from the USB port and remain on even though there's not enough power to power the stepper drivers.

This is kind of annoying.

You can power down individual USB ports on most hardware using `uhubctl`. 

It would be good to have this option from within the MainSail UI.

### Solves the following problem:

We don't have to either accept we'll have permanently semi-powered printers or faff about modifying USB cables.

meteyou, 2023-05-26:
thx for your request, but this feature have to be integrated in Moonraker first. Moonraker is the API and is running on the host itself. Mainsail is only a GUI and is only running in your browser on the client PC.

just my 2 cents: good boards have a jumper to disconnect the 5V from the usb interface. for bad boards, you can just cut the 5V wire in the usb cable. this is more a hardware issue of your printer. i think its not necessary to "fix" this with software.

hughesjs, 2023-06-04:
Ahh, my mistake, I'll raise an issue over there.

Although I disagree with your comment and I'll explain why.

1. Yes, it would be lovely it this was present on the board but it's not, and modifying cables is a pain.
1. A huge amount of features in Klipper are there to address hardware issues in software. Everything from bed meshing to twist compensation is there as a software fix for hardware issues.
1. It should also be a reasonably easy thing to do in software. The tool to control hub power already exists, it's not like we're going to have to worry about making awkward syscalls.

meteyou, 2023-06-16:
Software "hacks" are usually harder to set up for some non-linux users than a hardware mod. But if this feature will come in Moonraker, I will implement it in Mainsail.

Can i ask which board do you use?

Zeanon, 2023-06-22:
I don't know how important this still is, but instead of cutting the power you can just stick some electrical tape over the 5v pin in your usb-a plug, that's pretty easy

hughesjs, 2023-06-26:
@Zeanon - I know that you can... But when we can handle this in software, hardmodding a usb cable seems like something we should be avoiding

Zeanon, 2023-06-26:
Software is not possible, the rpis don't support to turn off individual USB ports
And I would hardly call putting electrical tape over a contact a hardware mod tbh XD

closed 2023-06-30 by meteyou