mainsail-crew/mainsail #1398: Let macros explicitly define display options

jschuh, 2023-05-26:
### Requested feature:

Not sure if this belongs under #484 or not, but it would solve a number of frustrations I've been dealing with when trying to integrate [my own macros](https://github.com/jschuh/klipper-macros) with front-ends. 

My suggestion is to add a scheme where a macro exposes it's front-end presentation as an explicitly named variable containing a dictionary of fields. For most macros this will just be a static dictionary that sets default values and input ranges. For more complex macros this dictionary can be updated in response to changes in printer state.

Here's a rough example of what I'm thinking:

```
[gcode_macro MY_MACRO]
variable_front_end_options: {
  'hidden': False, # Set to true to hide
  'params': {
      'MY_SELECT_ARG' { # Drobdown box
        'type': 'select'
        'select': ['choice_a', 'choice_b', 'choice_c']
        'default': 'choice_a'
      }
      'MY_NUMBER_ARG' { # Numeric entry field
        'type': 'number'
        'min': 0
        'max': 100
      }
      'MY_TEXT_ARG' { # String entry field
        'type': 'text'
        'max': 10
        'default': 'purple'
      }
    }
  }
# More macro stuff after...
```

The general idea is that before rendering the macro parameters pane, Mainsail would check for store.state.printer['front_end_options']; if it's not present it just falls back to what it inferred from parsing the config. If it is present, then it renders only the parameters in the dictionary, and according to the specified format.

In the case of MY_MACRO above the rendering would be as follows:

- MY_SELECT_ARG as a dropdown box with the default set to "choice_a"
- MY_NUMBER_ARG as a numeric value between 0 and 100 with no default
- MY_TEXT_ARG as a text entry field accepting up to 10 characters with a default of "purple"

### Solves the following problem:

The big things that this allows for are:

- parameters that determine their default values from printer state
- parameters that can have basic restrictions
- parameters that can be hidden from the UI
- hiding a macro without the underscore prefix (e.g. for macros that make sense in the console, but not as buttons)

meteyou, 2023-05-26:
i already thought of something similar, but how you managed it with 1 variable didn't occur to me yet. i find your suggestion very well realized. Thank you!

jschuh, 2023-05-26:
Thanks. If you look at my repo you'll find I've stretched the capability of macros far beyond what any reasonable person should (and I am not exactly proud of this 😉).

Anyway, I don't have experience with vue.js per se, but I am comfortable with JS/TS and Python generally. So, would you like me to take a crack at implementing it?

meteyou, 2023-05-27:
Would be nice. If you need help, feel free to open a PR in draft state and ask.

jschuh, 2023-05-27:
Okay. I set the dev environment up and got a proof of concept working yesterday. It's a holiday weekend here so I may not get back to it until Tuesday, but I'll polish it up a bit and post a PR when I do.