mainsail-crew/mainsail #1399: Allow pin/favorite gcode files.

GAZ082, 2023-05-26:
### Requested feature:

May times I print the same 20x20 sheet when I adjust my bed or want to see everything is fine, but its cumbersome to look for it in the file list. Would be nice to have an option to pin it and by doing so goes to the top of the list, like in Whatsapp conversations.

meteyou, 2023-05-26:
you only have to create a directory and put the gcode file in it. directories are always on the top.

GAZ082, 2023-05-27:
> you only have to create a directory and put the gcode file in it. directories are always on the top.

Nice workaround. But yet pinning would be nice.

meteyou, 2023-05-27:
To be honest, I see no advantage in pinning, compared to a directory.

GAZ082, 2023-05-27:
You will still want to have files on the top in that folder once the number of files increases.

meteyou, 2023-05-27:
Pinned files are also increasing and block space from other new files...

GAZ082, 2023-05-27:
> Pinned files are also increasing and block space from other new files...

If you overuse, yes. But you are free not to use.

I added a mockup to my first post.

meteyou, 2023-05-28:
Sry, but I don't see any advantage with pinned files compared to the folder. Rather only disadvantages, such as lost space in the width and blocked entries in the list.

th33xitus, 2023-05-28:
Honestly i would also just move the files i print regularly into a favorites folder, you can prefix it with `000_` for example and make sure it always stays at the very top (if you sort by name). It has the advantage that all your favorite files are in one place without blocking any space at all in the upper directory of the filebrowser. I see the ROI of that pinning feature quite low tbh.

GAZ082, 2023-05-28:
> Sry, but I don't see any advantage with pinned files compared to the folder. Rather only disadvantages, such as lost space in the width and blocked entries in the list.

If width is a concern, just add it as a right click option.

GAZ082, 2023-05-28:
> Honestly i would also just move the files i print regularly into a favorites folder, you can prefix it with `000_` for example and make sure it always stays at the very top (if you sort by name). It [...]

000_ is a hacky way, Mainsail is sleek.

forgodtosave, 2023-05-28:
Doesn't issue #1257 already exist for this?
I actually think this feature is a very nice idea.

I could try this. Should the pin be permanent or only available via the right-click menu?

meteyou, 2023-05-28:
@forgodtosave i would prefer it via right click and add the icon to the print status icon. so we don't waste space for smaller displays

meteyou, 2023-05-28:
but since the other FR is "older", i would close this issue, because it is a dublicate.

closed 2023-05-28 by meteyou