mainsail-crew/mainsail #1403: Bring back "touch" feature on touch-screens like BTT Pi TFT's

JTVybiral, 2023-05-29:
### Mainsail Version:

2.5.1

### Browser:

Chrome

### Device:

Laptop

### Operating System:

Windows

### What happened?

"touch" feature on Big Tree Tech's Pi TFT70 worked great for about a month, then suddenly quit. Suspecting that the problem was in the hardware, I replaced both the BTT TFT display AND the Raspberry Pi 3 B+ as I had spares of both. Still no "touch". The BTT TFT70 display is still crisp and very colorful.

### What did you expect to happen instead?

It would be great to have the "touch" feature restored, because the "Fine Tuning" button/controls only appear on the touch screen, and the GUI on the touch screen is very intuitive, quick and easy to use.

### How to reproduce this bug?

The "touch" feature quit working about a month ago. Otherwise, Mainsail, RatOS, Moonraker, Klipper, etc. all working great! The BTT TFT70 display is still working just fine, but without the "touch" feature.

meteyou, 2023-05-29:
This is an os issue and not an issue von Mainsail. Mainsail is just a webinterface. its not possible to disable "touch support" in a browser app.

closed 2023-05-29 by github-actions[bot]

JTVybiral, 2023-05-29:
Sorry to bother you. I'm just a beginner.

meteyou, 2023-05-29:
np. pls join our discord to get help fix your issue