mainsail-crew/mainsail #1404: Using webrtc video stream doesn't show on printers tab

smwoodward, 2023-05-30:
### Mainsail Version:

2.5.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

macOS

### What happened?

If webrtc is set as the service for camera's and you have more than 1 printer setup in mainsail, you are not able to select the camera stream on the printers tab.  It works properly if the camera's are set to MJEPG streaming.

### What did you expect to happen instead?

To be able to see the camera stream on the printers tab using webrtc just like MJEPG streaming works

### How to reproduce this bug?

Have more than 1 printer in Mainsail, have one printer's cameras set to MJEPG streaming and the other as webrtc.  Click on the printers tab after a refresh and you have the option to see the camera stream for the printer set as MJEPG but not the one for webrtc.

Henk72, 2023-06-02:
I have the same issue. I can get direct access from VLC to the rtsp stream, but not in the mainsail config for both cameras

appleimperio, 2023-06-04:
Same issue here. Printer setup with ustreamer shows up, the webrtc one no (only on the printers tab)

PR #1422 (2023-06-18): refactor: add webcam-wrapper component
## Description

This PR add a new component `WebcamWrapper` which can be used to mount every webcam service in each place without mounting all webcam services on each place.

## Related Tickets & Documents

This PR fixes #1404 

## Mobile & Desktop Screenshots/Recordings

none...

## [optional] Are there any post-deployment tasks we need to perform?
closed 2023-06-18 by meteyou