mainsail-crew/mainsail #1408: Large thumbnail dosent work with gcode thumbnail ?

PhilBaz, 2023-06-01:
### Mainsail Version:

2.5.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I don't get large thumbnail previews on gcode files with imbedded gcode thumbnails. But I do get them if the gcode file has a side-car image. Like when i choose 'UFP with Thumbnail' in connect moonraker

Just me?

[image]

### What did you expect to happen instead?

Same large thumbnail display on mouse over in gcode tab, or on the main page when printing.

### How to reproduce this bug?

Use gcode thumbnail type.

meteyou, 2023-06-01:
please read the docs: https://docs.mainsail.xyz/overview/slicer/cura

closed 2023-06-01 by github-actions[bot]

PhilBaz, 2023-06-01:
@meteyou Ahhh, the second thumbnail script....Sorry!!!