mainsail-crew/mainsail #1436: Fan adjust like flow adjust and speed.

JohnDCC, 2023-06-23:
### Requested feature:

An adjustment slider like the speed or flow ones that scales up part cooling fan (to a max of 100).

### Solves the following problem:

When fan settings aren't quite right, allows you to save a print mid print by adjusting up or down the cooling.

meteyou, 2023-06-24:
This is implemented since on of the first releases of Mainsail:

[image]

The first slider with the name `Fan` is your `[fan]`.

closed 2023-06-24 by meteyou

JohnDCC, 2023-06-25:
Yes this adjusts the current fan level but gets overwritten as soon as the gcode changes it to something else. What I am proposing is a multiplier that gets applied over the gcode.

meteyou, 2023-06-25:
Klipper has no option for that. So this needs first to integrated in Klipper.