mainsail-crew/mainsail #1438: Unable to edit a cfg file

kubaracek, 2023-06-24:
### Mainsail Version:

v2.6.0

### Browser:

Chrome

### Device:

Laptop

### Operating System:

macOS

### What happened?

A member of our group was initially complaining that on a fresh install he's unable to open either of his cfg files in Mainsail.
While I was just editing some klicky configs I noticed, that I couldn't open klicky-macros.cfg

At least for me it's easily reproducible, I tried to find what could be wrong in the sources, but couldn't really narrowed it down a single file. I reckon it could be introduced with #1200 as to me it looks like it's due to a syntax highlighting. 

It also happens on a larger cfgs, that initially do load, but after scrolling down, it errors with the same error and the editor is unusable. 

[image]

### What did you expect to happen instead?

To open the file and edit

### How to reproduce this bug?

create a `.cfg` file with

```
###################
# Klicky Dock Servo Deploy

[gcode_macro _DeployKlickyDock]
description: Deploys Klicky servo-controlled dock
gcode:
    {% set enable_dock_servo = printer["gcode_macro _User_Variables"].enable_dock_servo|default(False) %}
    {% set servo_delay = printer["gcode_macro _User_Variables"].servo_delay|default(1000) %}
    {% set servo_name = printer["gcode_macro _User_Variables"].servo_name %}
    {% set servo_deploy = printer["gcode_macro _User_Variables"].servo_deploy|default(360) %}

    #wait for all the moves to complete
    M400
    {% if enable_dock_servo != False %}
        _KlickyDebug msg="_DeployKlickyDock Klicky servo configuration enabled"
        {% if servo_deploy == 360 %}
            { action_raise_error("Klicky: servo active on klicky-variables, but no servo deploy angle specified") }
        {% endif %}
        _KlickyDebug msg="_DeployKlickyDock SET_SERVO SERVO={servo_name|string} ANGLE={servo_deploy|int}"
        SET_SERVO SERVO={servo_name|string} ANGLE={servo_deploy|int}
        M400
        G4 P{servo_delay|int}
        _KlickyDebug msg="_DeployKlickyDock SET_SERVO SERVO={servo_name|string} WIDTH=0"
        SET_SERVO SERVO={servo_name|string} WIDTH=0
    {% elif printer["gcode_macro _DeployDock"] is defined %}
        _KlickyDebug msg="_DeployKlickyDock calling _DeployDock"
        _DeployDock
    {% endif %}
```

kubaracek, 2023-06-24:
I tried to narrow it down and it breaks on this in the editor:

```
gcode:
    {%  %}
```

meteyou, 2023-06-24:
[image]
works without any issue with v2.6.0. So i cannot reproduce this issue.

closed 2023-06-24 by meteyou

kubaracek, 2023-06-24:
@meteyou It's a me and another guy who I never met having the same issue. The console output shows the error. I would keep it open in case someone else runs into that and maybe close due to inactivity. Either way, it's an issue. I can provide you with more info if any are requested.

meteyou, 2023-06-24:
@kubaracek sry. as long as i cannot reproduce it, i cannot fix it...

kubaracek, 2023-06-28:
And indeed, he mentioned it did not work for him in Safari (which I'm using) but works in Chrome.
So I checked in chrome and it works, so it's somehow Safari related.

meteyou, 2023-06-28:
@kubaracek pls upload your config file, that we can doublecheck it.

meteyou, 2023-06-28:
@kubaracek sry. i cannot reproduce it with safari:

[image]

MexicoSkies, 2023-07-21:
I also have this issue with safari. printer.cfg is "blued out" around line 275 and beyond which is not editable. Firefox and Chrome are working great. Safari Version 14.1.3 (16611.3.10.1.19) . Some other files did not even open when trying to edit - KAMP_Settings.cfg, mainsail.cfg (read-only). It would appear to be a Safari issue, which is a trend I am all too familiar with.

meteyou, 2023-07-21:
@MexicoSkies pls try to update your Safari. I have the Version 2.6.1 and it looks like, that Safari fixed this issue / is compatible with it.

MexicoSkies, 2023-07-21:
@meteyou. I did a system upgrade before my post this morning but, that didn't update Safari. I just went back to check for updates and lo and behold, there was a Safari update available. I am now running Safari 16.5.2 and everything is now editable in printer.cfg. It's great that these little nuisances come up because between 3d printing, software and hardware, hardly anything ever goes wrong! LOL. Thanks for your help!