mainsail-crew/mainsail #1444: Uneditable printer.cfg and no text populating in klicky-macros.cfg

Robbierob916, 2023-06-25:
### Mainsail Version:

v1.0.0-11-ge4976f6

### Browser:

Safari

### Device:

Desktop PC

### Operating System:

macOS

### What happened?

Updated Klipper, Mainsail and Moonraker and printer.cfg is no longer editable.  Also working on Klicky probe install and Klicky-macros.cfg will come up, but no text displayed.
[image]
[image]
[image]
[image]

### What did you expect to happen instead?

File to be editable and visible

### How to reproduce this bug?

Not sure.  Was working fine yesterday, prior to updates of Klipper, Moonraker and Mainsail.  For my case, SW, set up, it happens every time, does not clear after reboot.  I am running a Laptop with 2 instances of Klipper, moon raker.  The other printer.cfg displays and edits fine.  I can also update the files and view them from the Host laptop.

meteyou, 2023-06-25:
duplicate of #1433

meteyou, 2023-06-25:
and i think your mainsail version in your bugreport is also wrong. i think you mean v2.6.0

truckershitch, 2023-06-26:
@Robbierob916 Can you post your `printer.cfg` and `klicky-macros.cfg`?

I would like to look at them further.

meteyou, 2023-06-29:
@Mcrcheli if you speak from desktop pcs/macs, the browser + version number would be much more important, than the OS. so pls post your browser name and version number of your browser

Mcrcheli, 2023-06-30:
web browser: Safari
Version 16.3 (18614.4.6.1.6)

Robbierob916, 2023-06-30:
@truckershitch ...Sorry, got busy at work...Thanks for the ask...I will post both files, but forewarning, printer.cfg is a bit of a mess, because I was in the middle of installing/troubleshooting Klicky probe and deciding on if I wanted it servo actuated.  I kept getting a Jinja parsing error, so I updated Klipper, Mainsail and Moonraker.  After update, no bueno.  

As far as the Klicky-Macros.cfg, I would assume it will show fine in the post as I can read it and edit it outside of Mainsail.  The problem with Klicky-macros.cfg is that when I open it in Mainsail nothing is visible.  

[printercfg.txt](https://github.com/mainsail-crew/mainsail/files/11911469/printercfg.txt)

[klicky-macroscfg.txt](https://github.com/mainsail-crew/mainsail/files/11911470/klicky-macroscfg.txt)

meteyou, 2023-06-30:
sry. i tested both files with a fresh installed mainsail v2.6.0 and safari 16.5 (i think this is the newest release) without any issues. another user posted the same issue on the discord server and a clear cache of his browser fixed the issue for him. pls test this on your system.

closed 2023-06-30 by meteyou

Robbierob916, 2023-06-30:
Tests and results:
Mac mini: Safari 16.1, OS Ventura 13.0.1 - cleared Cache, no joy...
IMac: Safari 16.3, OS Monterey 12.6.5 (same issue) - cleared Cache, no joy...
Ipad OS 16.5.1, Safari Version??? - works, files visible and editable
Other tests:
Mac mini: OS Ventura 13.0.1, Google Chrome - works, files visible and editable
IMac: OS Monterey 12.6.5 Mac, Google Chrome - works, files visible and editable
Mac: Updated to Monterey 12.6.7, Safari 16.5 - cleared Cache, works, files visible and editable

Typing this on Mac mini, will try update and give feedback shortly...

Thanks for the help...

Robbierob916, 2023-06-30:
Final result, success... with update to MacOS 13.4.1 and Safari 16.5, clear cache - works and files are editable...

Thank you again

meteyou, 2023-07-01:
Thanks for all your tests! So it looks like a safari issue in older versions.