mainsail-crew/mainsail #1446: Spoolman filament manager integration

Donkie, 2023-06-26:
### Requested feature:

Moonraker will [soon](https://github.com/Arksine/moonraker/pull/651) support integration to my latest invention, [Spoolman](https://github.com/Donkie/Spoolman).

Front ends will be able to interact with the full [Spoolman API](https://donkie.github.io/Spoolman/) through a proxy available in Moonraker. So eventually, creating a full CRUD interface is entirely possible if so desired.

However, as a starting point, the most essential thing is being able to set/get the "active spool", which is the spool that Moonraker will track filament usage of to the Spoolman server. This can be done though the [web API](https://github.com/Arksine/moonraker/blob/364eef7481976e4a81a29b0a04745f77dd198eae/docs/web_api.md#spoolman-apis).

As a second step, displaying more information about the active spool would be cool, such as remaining filament, etc. Also perhaps having warnings pop up if the remaining filament isn't enough for a print before starting it?

meteyou, 2023-06-26:
Oh nice! This is a highly requested feature in the klipper world. Thank you for your work. I will ping you, if I have questions for the implementation or I have something to test.

patofoto, 2023-07-10:
Amazing!!!. Already running Spoolman on a Docker on my NAS. Soooooo looking forward to this integration. I am just a hobbyist good enough with computers to get in trouble by getting way too deep beyond my knowledge. I understand the concept of how everything is working but need to learn how to make API calls and display the results.

patofoto, 2023-07-10:
@Donkie Apologies if this is too basic a question. I am currently running Spoolman on my NAS. I assume I can install Docker directly on my control board and run Spoolman there?.  It is a BTT Manta E3EZ with a BTT CB1 and their latest image.

Donkie, 2023-07-10:
Never used that myself, but if it can run regular ARM/x86 programs then yes it should be possible.

Please use the mainsail discord in the future for any questions. Spoolman has it's own channel there, #spoolman

meganssmith, 2023-08-22:
any update on this?  Would love to help test

numindast, 2023-08-22:
> @Donkie Apologies if this is too basic a question. I am currently running Spoolman on my NAS. I assume I can install Docker directly on my control board and run Spoolman there?. It is a BTT Manta E3EZ with a BTT CB1 and their latest image.

This targets running on the host where Mainsail / Moonraker / Klipper are running. So this would be on a Raspberry Pi or other host computer you use.

nelsongraca, 2023-10-06:
Any predicted date on this?

PR #1542 (2023-10-06): feat: add spoolman support
## Description

This PR will add the support for [Spoolman](https://github.com/Donkie/Spoolman) ([Arksine/moonraker/pull/651](https://github.com/Arksine/moonraker/pull/651)).

## Related Tickets & Documents

fixes: #1446

## Mobile & Desktop Screenshots/Recordings

## [optional] Are there any post-deployment tasks we need to perform?
closed 2023-10-06 by meteyou