mainsail-crew/mainsail #1454: Config Subdirs are borken

kingspeedy95, 2023-07-05:
### Mainsail Version:

v2.6.0

### Browser:

Edge

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I have split my printer.cfg into several configs. I have split the new configs into subdirectories. But these subdirectories are not shown in the mainsail, only the whole content of the folders. When I try to open a config, Mainsail tries to download it. But can not open it. But no error message is displayed.

[image]
[image]
[image]
[image]
[image]

### What did you expect to happen instead?

I would have expected the subdirectories to be displayed in the same way as in the file system.

### How to reproduce this bug?

Split printer.cfg into several subdirectories.

meteyou, 2023-07-06:
If you move the files with an FTP program or similar fast, moonraker may not notice it in time, and the push messages between Moonraker and Mainsail are too slow. This should be solved with a simple refresh of the mainsail tab. (maybe a moonraker restart is necessary before) 

If you still have problems, please post the output of the following requests:

http://<IP or hostname>/server/files/list?root=config
http://<IP or hostname>/server/files/directory?path=config
http://<IP or hostname>/server/files/directory?path=config/configs

kingspeedy95, 2023-07-06:
Thanks for the quick response! I had already restarted the Pi once yesterday. After that it did not work. When I turned on the Pi today, everything worked as expected.

meteyou, 2023-07-06:
If you don't reload your browser tab, it didn't reload the directory from the pi to your browser. If you have unsync issues like that, a browser refresh should fix it.