mainsail-crew/mainsail #1461: ETA 12 Hour Shows Incorrect AM vs PM

fuzelet, 2023-07-13:
### Mainsail Version:

v2.6.0

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

Mainsail is set to use 12 hour timing, however the ETA is showing the incorrect time; AM vs PM.  Given this print only takes 2-3 HOURS, the ETA should say 12:18PM, not 12:18AM.

[image]

### What did you expect to happen instead?

ETA should be displaying 12:18PM, not 12:18AM based on current time and expected completion time (2-3 hours)

### How to reproduce this bug?

Use a file that runs long enough to change from AM to PM or PM to AM.

meteyou, 2023-07-13:
thx for reporting this bug. i think i will never understand the AM/PM time format...

fuzelet, 2023-11-23:
Not sure if this came back or not but I am seeing an issue with 12 hour clock and things ending around midnight.

Version 2.8.0

[image]

[image]

[image]

PR #1463 (2023-07-13): fix: fix issue with ETA and 12h time format
## Description

This PR will fix an issue with AM/PM switch at 12 PM in the StatusPanel ETA time

## Related Tickets & Documents

fixes #1461 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?