mainsail-crew/mainsail #1468: Slow website UI load times

jLynx, 2023-07-17:
### Mainsail Version:

Let me get back to you on that, but one of the latest 

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When I load the webpage it takes around 8-20 seconds to load. (Validated on multiple high-end devices). 

### What did you expect to happen instead?

The page to load a lot faster than it currently does

### How to reproduce this bug?

If you have a lot of history file, you will see this slow down. I just with a lot of use time (438hours) 

### Additional information:

hso the issue seems to be that it loads the entire history on page I it. My suggestion is to have it load the top X from history (maybe 10?) and the rest of the history can be loaded on the history page load 

https://github.com/Rat-OS/RatOS-configuration/assets/4393979/61c7380c-8376-4ade-ab3c-a7b693c39aa6

https://github.com/Rat-OS/RatOS-configuration/assets/4393979/4aa5a27d-b621-418f-bfd6-3ccf810838ad

zellneralex, 2023-07-17:
See also https://discord.com/channels/758059413700345988/1130299768661344288

jLynx, 2023-07-17:
After resetting my DB it still took ages to load.
After restoring the DB and then removing all my ``/home/pi/printer_data/gcodes`` files the page now loads contently in 3 seconds which is much better.

So that tells me that is still forcefully loading all the data in the ``gcodes`` folder on load

jLynx, 2023-07-18:
My OS is 64 Bit 
[image]

PR #1714 (2023-12-31): fix: fix long initial time with huge print history
## Description

This PR fix a issue with long initial times with huge print histories. It only loads the first 100 entries and the rest only when the user clicks on a button.

## Related Tickets & Documents

fixes #1468 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2023-12-31 by meteyou