mainsail-crew/mainsail #1481: Link to release notes on update interface

TexZeTech, 2023-07-24:
### Requested feature:

Have the version number link directly to git or custom source for release notes if it needs updates or not.

The ⓘ button can link to the commit change as Moonraker already does.

[image]

Thanks again for all the work.

### Solves the following problem:

Adds more convenience on the interface.to

meteyou, 2023-07-24:
do you see the update of mainsail 2 lines deeper? -> https://github.com/mainsail-crew/mainsail/issues/1481

closed 2023-07-24 by meteyou

TexZeTech, 2023-07-24:
I think we are thinking of two separate things. 

I would like to see the numbers let's say in this example mainsail v2.6.1 to then link to the release page like https://github.com/mainsail-crew/mainsail/releases/tag/v2.6.1 and the v2.6.0 would link to https://github.com/mainsail-crew/mainsail/releases/tag/v2.6.0 so that the user can just click on it and it would bring the information up. 

You could then expand on this and the current version could still link to the location it is pulling from.

Unless I'm truly blind that doesn't seem to be how it works.

TexZeTech, 2023-07-24:
I just re-read my initial post. I want the version number to open the git release information so I can read the changes.

meteyou, 2023-07-24:
@TexZeTech it's also my fault. i don't added all informations to the PR #1481. i found this "missing" function in the refactoring and linked the version number in this PR. So we talked about the same thing...

the only thing i did differnt is, that i only linked the new version (v2.6.1 in this case and not v2.6.0), because you have already the old version installed, so you only need the new informations.