mainsail-crew/mainsail #1485: mode: camera-streamer, WebRTC, set flip camera image level or flip, all are unresponsive.

skyaee, 2023-07-25:
### Mainsail Version:

v2.6.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

macOS

### What happened?

mode: camera-streamer, WebRTC, set flip camera image level or flip, all are unresponsive.
[image]
[image]

### What did you expect to happen instead?

Upgrade from v3 to v4

### How to reproduce this bug?

After reinstalling the system on Raspberry Pi

meteyou, 2023-07-25:
thanks for the report! the bug was bigger than expected. it affected almost all webcam clients.

closed 2023-07-27 by meteyou

R4toLoko, 2023-07-30:
The timelapse appears to have been affected also.
It usually shows a thumbnail of the last shot it took, right now it shows nothing like if the time-lapse was disabled, then it shows the recording at the end.
So atm you don't really now if it is running or not until the end.

I have forgotten to enable it in the past and this timelapse thumbnail helped to identify it, stop the print, start it again and voila.

R4toLoko, 2023-07-31:
@meteyou this whole problem appears to be crowsnest doing, not mainsail.
I am still collecting more info but it seems like they release a poop update.

meteyou, 2023-07-31:
@R4toLoko plus don't use other issues for different problems. Pls open a new issue, if you think there is a bug!

meteyou, 2023-07-31:
@R4toLoko Yes, such things are important, and for this reason, issues should be separated so as not to be forgotten. Therefore, I ask you to create a new issue so that it will not be forgotten. This is to keep the overview.

PR #1487 (2023-07-27): fix: fix flip funciton in several webcam clients

## Description

This PR fix the flip function in all webcam clients except MjpegstreamerAdaptive.

## Related Tickets & Documents

fixes #1485 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none