mainsail-crew/mainsail #1490: Filament cannot be extruded, and the extrusion button is gray and cannot be pressed.

skyaee, 2023-07-27:
### Mainsail Version:

v2.6.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

https://github.com/mainsail-crew/mainsail/assets/40803751/cec9e3ff-d922-46e5-818c-c41c141eab23

Filament cannot be extruded, and the extrusion button is gray and cannot be pressed.

### What did you expect to happen instead?

Filament cannot be extruded, and the extrusion button is gray and cannot be pressed.

### How to reproduce this bug?

Filament cannot be extruded, and the extrusion button is gray and cannot be pressed.

BlackShadeOSS, 2023-07-28:
It happened to me a lot of times, you just need to refresh browser to fix it. It happens when you do firmware restart when the printer is hot.

skyaee, 2023-07-28:
> 这种情况我遇到过很多次了，只要刷新一下浏览器就可以解决。当打印机很热时重新启动固件时会发生这种情况。

meteyou, 2023-07-29:
@skyaee thank you for the tip with the reset after a high extruder heater. pls write here only in english... i will reopen this bug report and try to fix it.

meteyou, 2023-07-29:
Ok. I found a bug in the API init process. After a firmware_restart/restart it always initialize the extruder>can_extrude value as false and don't update it after that, only you pass the min_extrude_temp. So i added a "hack" to reinit this value after a short delay.

A real fix of this, have to be implemented in Moonraker/Klipper, bit it would work fine with my fix in Mainsail.

PR #1495 (2023-07-30): fix: fix issue with extruder cannot extrude after a klipper restart
## Description

After a Klipper firmware_restart/restart and the extruder temp is higher than the min_extrude setting, Moonraker/Klipper only init the extruder with `can_extrude: false` but don't update it. With this PR, i send a delayed reinit request to update this attribute. This is more a "hack" then a fix, because the fix would have to be in Moonraker/Klipper itself.

## Related Tickets & Documents

fixes #1490 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to [... cut]
closed 2023-07-30 by meteyou