mainsail-crew/mainsail #1493: Editing a webcam via Mainsail breaks Moonraker's webcam API

Clon1998, 2023-07-28:
### Mainsail Version:

2.6.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

After updating to 2.6.1 mainsail breaks the webcam API by writing int values instead of boolean values for the flipX and flipY fields to the moonraker API. 
As seen in both the [examples](https://moonraker.readthedocs.io/en/latest/web_api/#list-webcams) and the [code](https://github.com/Arksine/moonraker/blob/937c766ded8213d43ff58bde5355e31e13e35b88/moonraker/components/webcam.py#L412-L413) of moonraker a boolean is expected, obviously in that case some validation on moonraker's side is also missing to prevent this in the future.

### What did you expect to happen instead?

Mainsail should also use bool when writing the values back to moonraker.

### How to reproduce this bug?

Just add or edit a flipX/flipY webcam.

meteyou, 2023-07-28:
I posted it already in the mainsail support ticked. Mainsail sends "true/false" but Moonraker send a "1/0" back. So Moonraker is mixing up this value, when you use the API to update the webcam. When I write directly in the DB, it works.

closed 2023-07-28 by meteyou

Clon1998, 2023-07-28:
> I posted it already in the mainsail support ticked. Mainsail sends "true/false" but Moonraker send a "1/0" back. So Moonraker is mixing up this value, when you use the API to update the webcam. When I write directly in the DB, it works.

Interesting, you got a link to the report on Moonraker?

meteyou, 2023-07-28:
here is the link to the support ticket: https://discord.com/channels/758059413700345988/1133600080864219146/1133705481119092856

you can see in the screenshot the request from Mainsail and it sends true/false to Moonraker. after this request, i checked the api list output and it was 1/0.

But i see some other issues in your output above. There are many outdated values in your output above. Here is a link to the Moonraker docs, where you can see all new attributes: https://moonraker.readthedocs.io/en/latest/web_api/#webcam-apis
these values should be updated, when you edit a webcam in Mainsail.

Clon1998, 2023-07-28:
> here is the link to the support ticket: https://discord.com/channels/758059413700345988/1133600080864219146/1133705481119092856
>
>
>
> you can see in the screenshot the request from Mainsail and it [...]

Thanks for the heads up.
I already migrated to the Api and those values might be left from older version I will verify that.