mainsail-crew/mainsail #1494: After the latest update, webcam is shown upside down

R4toLoko, 2023-07-29:
### Mainsail Version:

v2.6.1

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

After the last update, well, I only updated everything last night, my webcam is upside down.
I have made no changes in my end, only the the updates.
I have those options enabled since day one because the cam is mounted upside down. 
It was never an issue.

[image]

[image]

### What did you expect to happen instead?

The camera to be working as expected.

### How to reproduce this bug?

I just turned the printer on and the cam is upside down with no way to fix it.

R4toLoko, 2023-07-29:
That is weird, the timelapse looks correct

[image]

meteyou, 2023-07-29:
Duplicate of https://github.com/mainsail-crew/mainsail/issues/1485 and already fixed. Plus check first, if there was already a bug report, before opening a new one!

closed 2023-07-29 by meteyou