mainsail-crew/mainsail #1505: Internal error during connect: float division by zero

DarthKarnagge, 2023-08-02:
### Mainsail Version:

v1.0.0-16-gc64d3af

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When I try to connect to a Tronxy X5SA with a v10 motherboard via USB in mainsail, I get the error: Internal error during connect: float division by zero.

### What did you expect to happen instead?

Connect to my printer.

### How to reproduce this bug?

Install klipper on the motherboard and connect with a USB

### Additional information:

Here is the relevant section of the klippy log

=======================
Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
webhooks client 1965328128: New connection
webhooks client 1965328128: Client info {'program': 'Moonraker', 'version': 'v0.8.0-136-g797c623'}
Loaded MCU 'mcu' 119 commands (v0.11.0-257-ged66982b / gcc: (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027] binutils: (2.34-4+rpi1+14) 2.34)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c1a=PB8,PB9 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_i2c3=PA8,PC9 BUS_PINS_sdio=PC12,PD2,PC8,PC9,PC10,PC11 BUS_PINS_spi1=PA6,PA7,PA5 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_spi3=PB4,PB5,PB3 BUS_PINS_spi3a=PC11,PC12,PC10 BUS_PINS_spi4=PE13,PE14,PE12 CLOCK_FREQ=180000000 MCU=stm32f446xx PWM_MAX=255 RECEIVE_WINDOW=192 RESERVE_PINS_crystal=PH0,PH1 RESERVE_PINS_serial=PA10,PA9 SERIAL_BAUD=250000 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 176, in _connect
    self.send_event("klippy:mcu_identify")
  File "/home/pi/klipper/klippy/klippy.py", line 263, in send_event
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/klippy.py", line 263, in <listcomp>
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/extras/temperature_mcu.py", line 82, in _mcu_identify
    func()
  File "/home/pi/klipper/klippy/extras/temperature_mcu.py", line 137, in config_stm32f4
    self.slope = (110. - 30.) / (cal_adc_110 - cal_adc_30)
ZeroDivisionError: float division by zero

DarthKarnagge, 2023-08-02:
Update!

Fixed. It was a conflict with the Motherboard temp sensor and I commented out that section in printer.cfg

[temperature_sensor mcu_temp]
sensor_type = temperature_mcu
min_temp = 0
max_temp = 100

meteyou, 2023-08-02:
This is a Klipper error and no mainsail error. Mainsail is only a GUI.

closed 2023-08-02 by meteyou